CaringCaribou/caringcaribou #7: DCM discovery issue

bensya, 2016-05-15:
I just tested a car yesterday, run "python cc.py dcm discovery""

As the car I tested that it will generate periodic can message 0X12D, and msg.data[1]=50, I got almost every airID that support services.

It's normal response after checking the source in the below:

      def response_analyser(msg):
                # Catch both ok and negative response
                if msg.data[1] in [0x50, 0x7F]:
                    print("\nFound diagnostics at arbitration ID 0x{0:04x}, "
                          "reply at 0x{1:04x}".format(arb_id, msg.arbitration_id))
                    can_wrap.bruteforce_stop()
            return response_analyser

kasperkarlsson, 2017-05-22:
This is obviously a problem, but I don't see an obvious workaround. One could

1. Perform an extra attempt against all identified hits, in order to verify that they respond every time. Remove false positives that only result in responses to the first attempt.
2. Add an initializing step, which listens for messages matching the catch criteria _before_ starting the bruteforce. Blacklist the messages received based on e.g. arbitration ID, so that they will not be caught as successful responses.

None of these methods would be completely foolproof. However, adding an optional module argument which reduces noise by enabling method 2 (or even a combination of both methods) would be a useful improvement.

kasperkarlsson, 2017-07-09:
Added a -blacklist option to DCM discovery in https://github.com/CaringCaribou/caringcaribou/commit/37b72c1e127a8035b29ddc7221b29610ffe78c64. Individual arbitration IDs can be blacklisted, so that responses on those IDs are ignored.

Example: in order to ignore responses on arbitration ID 0x123 and 0x77c:
`./cc.py dcm discovery -blacklist 0x123 0x77c`

kasperkarlsson, 2017-07-09:
Added a `-autoblacklist N` to DCM discovery in https://github.com/CaringCaribou/caringcaribou/commit/b4fb75821fc876ad563eda71bbb6800def38cac7 as suggested in point `2` above. This flag makes the module (passively) listen for `N` seconds before running the discovery. All arbitration IDs that have sent seemingly valid DCM responses are automatically added to the blacklist. Of course, this can be combined with a manual blacklist through `-blacklist`.

Example usage:
1. `./cc.py dcm discovery -autoblacklist 10` Blacklist all false positives found in 10 seconds before starting the discovery bruteforce
2. `./cc.py dcm discovery -autoblacklist 60 -blacklist 0x123 0x77c` Blacklist 0x123, 0x77c and all false positives found in 60 seconds before the discovery bruteforce

This issue is now considered resolved.

closed 2017-07-09 by kasperkarlsson