CaringCaribou/caringcaribou #11: Send list of messages from file

kasperkarlsson, 2017-06-12:
Add a replay feature for an existing CAN dump, so that pre-defined lists of messages can easily be sent without adding them as command line arguments. This would e.g. be useful for repeatable exploits and automated testing.

It would make sense to add this as a `-f <filename>` flag to /tool/modules/send.py

Some reasonable input formats are
1. Default /tool/modules/send.py argument format
2. python-can (default dump format for /tool/modules/dump.py)
3. candump (as produced by /tool/modules/dump.py with `-c` flag).

mjidhage, 2017-06-20:
Blind sending after time x? 
I think we also need to "wait for this response pattern and then send the next line"
A variant is to create a request-response-pair (if CC can read this on the bus, then send this).

kasperkarlsson, 2017-06-21:
@mjidhage "Blind sending after time x" - do you mean that we should add a (configurable) time offset between each message? That sounds very reasonable to me.

Pattern matching: interesting idea, but code like that can already be written in python as a CaringCaribou module. Do we really wish to develop a new scripting language under the context of CC?

The idea of a CAN-based version of `expect` is of course interesting. I do however believe that such a project should rather be developed in a separate repository.

kasperkarlsson, 2017-07-06:
Replay feature for candump format implemented in https://github.com/CaringCaribou/caringcaribou/commit/a6773bf349018aa79e385253934dad133b7effe4

kasperkarlsson, 2017-07-06:
Left to do before closing:
1. Add support for default pythoncan format
2. Update documentation

I aim to take care of these actions on Sunday.

kasperkarlsson, 2017-07-09:
Support for pythoncan format added in https://github.com/CaringCaribou/caringcaribou/commit/08b1efbd78a2467994b9a88cd2bb263d4d8bad6a
Small fix - support for hexadecimal arbitration IDs for candump format in https://github.com/CaringCaribou/caringcaribou/commit/21f206922db9aeca26056e68849843347d6fa851

closed 2017-07-09 by kasperkarlsson