CaringCaribou/caringcaribou #12: Security access seed randomness analysis

kasperkarlsson, 2017-06-20:
Implement a module which analyses randomness of seeds for security access. Data can be gathered by sending a (large) number of Request Seed [0x27 0x01] messages.

What about ECUs which only renew their seed on boot? Maybe a renewal can be triggered by responding to the seed with an invalid key? Any other ideas?

bhass1, 2018-03-07:
+1 this feature. Would love to help when I get some more time and the weather gets nice enough for me to sit in my car.

To handle ECUs that only renew their seeds on boot (which is becoming much more common), you could send a UDS ECUReset message to cause a hard-reset. I believe the payload is: 02 11 01 00 00 00 00 00.

Another sub-feature I would like to see is attempting to send a random key (Send Key) in response to the Request Seed response. It's a simple pen-test to check if the keys are actually being checked. Probably low priority though because most ECUs will have an actual check in place.

kasperkarlsson, 2018-03-08:
@bhass1 These ideas sound great and your help would of course be welcome. If you do implement a module, just send a pull request and we will look into it!

If you happen to find specific use cases which are only valid for a certain vendor or product line/vehicle series, this could also be of interest. While this enhancement proposal issue considers a generic module, a vendor specific analysis module could also prove useful to other CC users.

bhass1, 2019-02-12:
FYI, I have code to perform seed dump in a private branch. It has been tested on a bench module. Waiting for company approval to submit.

I'm working on a few other features that I hope to open issues for and submit.. eventually :).

kasperkarlsson, 2019-02-14:
That sounds great, @bhass1! I'll keep my fingers crossed for your company's approval of sending a PR 👍

Cr0wTom, 2021-10-30:
#70 pull request regarding seed randomness fuzzer for ECUs where the random number generator is seeded with the system timer. :)

kasperkarlsson, 2021-11-25:
Implemented in #70.

closed 2021-11-25 by kasperkarlsson