CaringCaribou/caringcaribou #21: Implement arbitration ID blacklisting in XCP discovery

kasperkarlsson, 2017-07-10:
Same as https://github.com/CaringCaribou/caringcaribou/issues/7, but for XCP. We need a way to blacklist responses from individual arbitration IDs from command line. It would also be nice with a passive scan for automatic blacklisting before starting the discovery.

It would make sense to add a blacklisting engine to _can_actions.py_ and use this for both DCM and XCP.

mjidhage, 2018-09-28:
Came across a real world scenario the other day when this was a show-stopper. Some arbid were sending measurement-signals looking like XCP responses making all ArbID XCP requests positive :(

The possibility to blacklist some specific ArbId's would solve the problem.

kasperkarlsson, 2018-10-02:
Both static blacklisting and automatic blacklist scan have been implemented in the following commits:
https://github.com/CaringCaribou/caringcaribou/commit/810174fd328da973dbf49ae0dcd85ca133c5f732
https://github.com/CaringCaribou/caringcaribou/commit/bb4c27d4420617da1dc58e8d2fd64779ccee4c26
https://github.com/CaringCaribou/caringcaribou/commit/8e4ed68e66bd686da604182b5dc87f4b2e8ab2d6

`./cc.py xcp discovery -blacklist 0x123 0xabc` Run discovery, but ignore responses using arbitration ID 0x123 or 0xabc (you can provide as many as you like)
`./cc.py xcp discovery -autoblacklist 10` Passively listen to the CAN bus for 10 seconds and blacklist all arbitration IDs where messages which could be misinterpreted as XCP responses are found. After this, automatically run a scan using this blacklist.

The `-blacklist` and `-autoblacklist` options can of course be combined.

@mjidhage Feel free to give it a try! :-)

closed 2018-10-02 by kasperkarlsson