CaringCaribou/caringcaribou #32: Per-session log file

Aezore, 2018-06-22:
I suggest adding an appended log recorder, like a blackbox with datetimed results and user actions, with an user-setting for "how big" the log should be and trim it if needed. So in case you need to review some data or behaviour not within the session but another day or week even..as far as the log file size is set, in plain human readable text. This might be helpful as it really comes in handy sometimes on my Dataman chip programmer to actually double check if I did program a chip in the past and what data was loaded into.

kasperkarlsson, 2018-06-24:
This sounds like a nice feature, which could be really useful for users who make many or longer runs.

Logging e.g. script arguments, start time, end time, result summary, reason for quitting etc would give a nice, basic event log for each session. This could even be controlled by a log level value in a configuration file, so that the behavior can be persistently controlled without having to pass a log level argument each time CC is run. This would also be a reasonable place to set the maximum log size.

I like the idea. Any additional thoughts or input on this proposal are of course welcome.

kasperkarlsson, 2018-06-29:
Since `.canrc` is not specific to Caring Caribou (but rather belongs to the underlying python-can library), we should not be modifying its contents. I would prefer a configuration file located within the project itself.

It would make sense to utilize the `logging` python module to handle different log levels. This way, the setting from the configuration file could simply be used to initiate a call with a matching level to `logging.setLevel`