CaringCaribou/caringcaribou #39: Dump not complete when using python3

timower, 2018-10-08:
When dumping the CAN bus using the dump module to either a file or just stdout some packets are missing when using python 3. 
I tested using a virtual can bus and the [ICSim](https://github.com/zombieCraig/ICSim) application.

When changing the python interpreter to python 2 all packets are logged but when using python 3 some packets (such as the blinker controls for example) are missed.

kasperkarlsson, 2018-10-09:
@timower Do you mean that certain (types of) messages never get logged?

Can you please share some examples of how these blinker control messages look (arbitration ID, payload data, flags etc)? CAN dumps of such a session from both Python 2 and 3 would also be interesting, of course.

kasperkarlsson, 2018-10-29:
@timower I have changed the structure of the dump module, in order to significantly decrease CPU load. Can you please check if this resolves your issue?

Changes introduced in https://github.com/CaringCaribou/caringcaribou/commit/59a9bee4e1d07664317ec472f4c6fc3cafddc6dc

kasperkarlsson, 2018-11-06:
@timower Almost one month has passed without response, so I am still unable to reproduce the issue.

If you intend to verify the patch, share logs etc but haven't had time to do so, please let me know. Otherwise this issue will be closed shortly.

timower, 2018-11-06:
Sorry for the late answer, the issue can be closed.
I tested on the master branch and I found all packets.

closed 2018-11-06 by timower