CaringCaribou/caringcaribou #47: How to read ECU EEROM via OBD/CAN?

brandonros, 2018-11-28:
I apologize in advance that this question isn't specific to this project.

How are tuning manufacturers able to download the EEPROM from the ECU via OBD-II, tune it, then reupload it? For example, do Bosch ECUs have a special proprietary command that is not documented anywhere to send the ECU into a boot mode?

Any information you can share would be helpful.

enok542, 2018-11-30:
I do not know how to. but they are probably written in .dll to kess v2

kasperkarlsson, 2018-11-30:
I am sorry, but the issues section of this repository is not a general discussion forum.

closed 2018-11-30 by kasperkarlsson

SecureCloud-biz, 2023-06-18:
but it is a valid question, that wasn't answered ?? are there commands to deal directly with the ECU.. ex.. read the EEPROM ? Write it back...