CaringCaribou/caringcaribou #55: NotImplementedError: Invalid CAN Bus Type - None

icecolor-wiattack, 2019-04-15:
pyhton vsersion: python-2.7

python-can vsersion: python-can (3.1.1)

example:
sudo python cc.py dcm discovery -min 0x700  ()

-------------------
CARING CARIBOU v0.1
-------------------

Loaded module 'dcm'

Traceback (most recent call last):
  File "cc.py", line 114, in <module>
    main()
  File "cc.py", line 106, in main
    mod.module_main(args.module_args)
  File "modules/dcm.py", line 417, in module_main
    args.func(args)
  File "modules/dcm.py", line 199, in dcm_discovery
    with CanActions() as can_wrap:
  File "/home/qili/Desktop/AAAAA/caringcaribou/tool/can_actions.py", line 64, in __init__
    self.bus = can.interface.Bus()
  File "build/bdist.linux-x86_64/egg/can/interface.py", line 108, in __new__
  File "build/bdist.linux-x86_64/egg/can/util.py", line 197, in load_config
NotImplementedError: Invalid CAN Bus Type - None

I used all the modules, it all prompted NotImplementedError: Invalid CAN Bus Type - None

I tried to install the old version of the python-can module, but still can't solve my problem. Please help me answer, thank you very much. I am very anxious. Thank you!

closed 2019-04-23 by icecolor-wiattack

kasperkarlsson, 2019-05-02:
Hello @icecolor-wiattack and sorry for the late response.

Did you find out what caused the problem in your setup?

rkollataj, 2019-11-28:
@kasperkarlsson I had exactly the same problem. It seems that you are not configuring "interface" for python-can. To workaround it I have created .canrc file as below:

$ cat ~/.canrc 
[default]
interface = socketcan

----------------------------------------------------

$ python --version
Python 3.8.0

$ pip3 show python-can
Name: python-can
Version: 3.3.2
Summary: Controller Area Network interface module for Python
Home-page: https://github.com/hardbyte/python-can
Author: Brian Thorne
Author-email: brian@thorne.link
License: LGPL v3
Location: /usr/lib/python3.8/site-packages
Requires: wrapt, aenum
Required-by:

kasperkarlsson, 2020-03-24:
If anyone else is having the same problem: the need for a `.canrc` file is documented in the installation guide.

https://github.com/CaringCaribou/caringcaribou/blob/master/documentation/howtoinstall.md