CaringCaribou/caringcaribou #61: Well, apparently this repo is dead, soooooo

RoninDusette, 2020-05-30:
With all due respect, you guys built something amazing, and not maintaining it is simply disgraceful. I feel like I am just going to work off of my fork and not pass upstream, considering nobody here is paying attention. :/ Really sad.

kasperkarlsson, 2020-06-07:
No one has had access to the ECU lab and test rigs used for development and verification of CC in many months, due to the current COVID-19 situation. Trust me when I say that you are not the only one who is frustrated about the situation.

I am sorry to hear that you find my behavior disgraceful, but harsh words are unlikely to grant me access back to a temporarily closed down research environment.

Have a nice day.

closed 2020-06-07 by kasperkarlsson

RoninDusette, 2020-06-07:
Well, I was not aware that you guys had an actual testing lab with that
type of equipment. Also didn't see anything that indicated as such. I
didn't think my words were harsh. Just seems like abandoning something like
this is foolish, but now that I see the reason (would probably help to
update the README, but I am probably one of the few that cared, so I guess
it doesn't matter), I understand.
----------------------
*Ronin Dusette*
www.dusette.net