CaringCaribou/caringcaribou #69: Delay between messages - UDS

Cr0wTom, 2021-09-12:
I am testing a device (blackbox testing) and the result from the UDS service enumeration script is giving me 255/255 as the enabled services. 

I know in fact that in the device not every service is open. So my guess is that there is some kind of security mechanism checking for the delay between the UDS messages sent consecutively. 

Could this be the issue, and if yes, can the feature be implemented in a future update?

kasperkarlsson, 2021-09-13:
Hello @Cr0wTom ,

It might be that your device responds to all service requests, with another value than `SERVICE_NOT_SUPPORTED`. This logic can be seen here https://github.com/CaringCaribou/caringcaribou/blob/f8f09750c7aca39f63ec33063d4406f6dc78436a/tool/modules/uds.py#L354-L364

Can you please supply communication logs for the communication on the `src` and `dst` arbitration IDs included in your scan? Without knowing how your device is responding, it is hard to see why you get certain results in your environment.

This can be done by running `cc.py dump -f output.txt ARB_ID_1 ARB_ID_2`, where you replace `ARB_ID_1` and `ARB_ID_2` with the two arbitration IDs (`src` and `dst`), respectively. Keep this logging on while running the UDS service enumeration to capture the traffic.

kasperkarlsson, 2023-03-20:
Closing old, inactive issue.