CaringCaribou/caringcaribou #72: Support for UDS over DoIP

Cr0wTom, 2022-01-20:
Hello guys,

A lot of manufacturers start using this implementation. It will be good to see if first of all it is applicable and if we can implement it in the tool. It will be a nice addition.

kasperkarlsson, 2022-01-21:
Hello @Cr0wTom,

This is indeed interesting, but I must admit that I have only _read_ about this standard, and never actually tinkered with it myself. I do not know it well enough to estimate how hard it would be to add it to the `UDS` module (or perhaps as an alternative way to initialize `IsoTP` in `caringcaribou/tool/lib/iso15765_2.py`?)

Do you have any ideas about how this could be implemented in CC? By pulling in an existing "pretty much `python-can` compatible" DoIP implementation, or writing a new one? I guess it would make sense if one can simply choose another transport layer than CAN when using ISO-TP, but I am open to any suggestions.

To summarize: I am not very familiar with this, but if we can find a way to get this into CC that would be great!

Cr0wTom, 2022-01-21:
Hello @kasperkarlsson 

Yes i also hope to be as simple as choosing another transport layer.

I had some preliminary analysis and now i have a device in my hands (under NDA unfortunately) so i will start developing the tooling as i need to test the device either way!

I'll keep this updated and I'll come back with a pull request if i succeed :)

kasperkarlsson, 2022-01-21:
That sounds great! We'll wait and see if your research leads to any promising results which can be shared, then.

Your engagement is, as always, very appreciated 😃

Cr0wTom, 2022-02-15:
Hello @kasperkarlsson , I build the script using doipclient and udsoncan libraries as it was the "easy" way forward. 

In the previous modules of cc, I can see that you are using custom libraries for the different ISOs that I don't have the time to implement. 

Please let me know if I should create a pull request or if I should keep my fork for now, as a usable alternative with DoIP functionality :)

kasperkarlsson, 2022-02-15:
That sounds awesome, @Cr0wTom!

I did indeed implement some ISOs on my own, when I couldn't find solid implementations with suitable licenses at the time. Using existing libraries instead (whenever possible) is great 👍 A PR would be most welcome!

Cr0wTom, 2022-03-24:
Hello, 

A bit late, but I managed in the end. You can find the PR here https://github.com/CaringCaribou/caringcaribou/pull/73

Unfortunately, I will not have access anymore to the DoIP unit, so from now on I will not be able to fix any potential issues. All the functions of the module are working though. :)

kasperkarlsson, 2022-03-24:
That's great, @Cr0wTom ! I will check this out as soon as I can find a suitable time slot (so don't worry if it takes a few days) 👍

kasperkarlsson, 2022-04-01:
Implemented in #73.

closed 2022-04-01 by kasperkarlsson