CaringCaribou/caringcaribou #74: Bruteforce security access key and security levels

jnsbal, 2022-04-01:
Hello,

I am currently getting familiar with car hacking and found your tool.

Since I am not in expert in the field I got two questions regarding possible additions to CC.

Would it be an appropriate use case for caringcaribou to add a function to bruteforce the key on Security Access after the seed request?
I also wondered if it makes sense to try all possible values for the security level (related to cc.py uds security_seed) when checking an ECU.

Thanks!

kasperkarlsson, 2022-05-02:
Hello,

Bruteforcing different key values after a seed request would absolutely be possible. Another similar mode would be to also send a new seed request before each attempt.

As for the security level, I suppose that could be bruteforced as well 😃

bhass1, 2022-12-09:
Hey @jnsbal! I had started this work and skeleton implementation, but didn't make time in my life to fully implement and merge it with the mainline. Here it is for reference: https://github.com/bhass1/caringcaribou/tree/master_security_access

Feel free to open a PR on my fork if you want to take that approach or reference it in your own implementation.

obbardc, 2024-06-22:
Can you open a PR for that @CanBusHack / @bhass1 ?