CaringCaribou/caringcaribou #75: missed files

daytona675, 2022-04-22:
I'm trying to run DoiP module and it doen't work.
From the Doip.py, there is somes import (import DoipClient and import udsonscan) but the python file corresponding to these import weren't present in the repo.
Is it possible to push again theses files ? 
King regards.

kasperkarlsson, 2022-04-22:
These are external packages available from PyPI, i.e. they are not developed as part of Caring Caribou.

https://pypi.org/project/doipclient/
https://pypi.org/project/udsoncan/

You can install them through `pip` 😃

closed 2022-04-22 by kasperkarlsson

daytona675, 2022-04-22:
Many thanks !
Is it possible to write it on the main readme or on other part of the git ?
Have a Nice week end
King regards

kasperkarlsson, 2022-04-22:
That's a great idea which will hopefully help others as well 👍 I added a note on this in the module documentation: https://github.com/CaringCaribou/caringcaribou/commit/b664536f28f34afec1bdefc8bbf60d01f64f3649

Thanks!

daytona675, 2022-04-22:
> note on this in the module documentation: b664536

You re welcome ☺️