CaringCaribou/caringcaribou #77: inconsistent use of padding in xcp module.

LukasMarckmiller, 2023-01-31:
Is there a reason why the padding is omitted for xcp connect in line:
https://github.com/CaringCaribou/caringcaribou/blob/b664536f28f34afec1bdefc8bbf60d01f64f3649/tool/modules/xcp.py#L471

but not for other xcp subfunctions like the command discovery:
https://github.com/CaringCaribou/caringcaribou/blob/b664536f28f34afec1bdefc8bbf60d01f64f3649/tool/modules/xcp.py#L241-L246

Without padding the ECU receives frames that are not consistently 8 bytes which leads to problems parsing the frame. 

Suggested fix: Consistent use of padding and change
https://github.com/CaringCaribou/caringcaribou/blob/b664536f28f34afec1bdefc8bbf60d01f64f3649/tool/modules/xcp.py#L471

to 
```python
can_wrap.send_single_message_with_callback([0xff, 0, 0, 0, 0, 0, 0, 0], handle_connect_reply)
```

LukasMarckmiller, 2023-01-31:
I'll wait for feedback before submitting a PR.

kasperkarlsson, 2023-02-02:
Great catch, a PR would be most welcome 👍