CaringCaribou/caringcaribou #81: Attempted relative import in non-package

HeartSleep, 2023-03-22:
Traceback (most recent call last):
  File "caringcaribou.py", line 7, in <module>
    from .utils import can_actions
ValueError: Attempted relative import in non-package

HeartSleep, 2023-03-22:
Traceback (most recent call last):
  File "caringcaribou.py", line 7, in <module>
    from .utils import can_actions
ImportError: attempted relative import with no known parent package

HeartSleep, 2023-03-24:
I followed the new installation instructions to install the tool, but I am unable to call it in the shell.What other methods are there?

HeartSleep, 2023-03-24:
Hello @kasperkarlsson.
Thanks! I installed it according to the new installation method, running it directly in the shell found that it would run successfully.

kasperkarlsson, 2023-03-24:
That is great to hear. I will close this issue then, thanks for using Caring Caribou 😺

closed 2023-03-24 by kasperkarlsson