CaringCaribou/caringcaribou #82: USD security seed request does not handle PENDING response correctly

vdechef, 2023-04-05:
I try to use caring caribou to generate a batch of UDS Security Seeds. On some ECUs it works fine, but on others I get a NRC ` REQUEST_CORRECTLY_RECEIVED_RESPONSE_PENDING (0x78)` and Caring Caribou just stops afterwards.

After a bit of digging in the code, the problem comes from [this line in iso14229_1.py](https://github.com/CaringCaribou/caringcaribou/blob/547d24250f5106752a704b2510e9f8ac5dc1749f/caringcaribou/utils/iso14229_1.py#L296). The length comparison is strict, whereas it should allow equality.

I will make a PR to fix this.

kasperkarlsson, 2023-04-05:
Hello,

Very nice catch - it does indeed look like an edge case bug. A PR would be most welcome - if you do this, feel free to add your name to the `Contributors` section of `README.md` as well 👍

kasperkarlsson, 2023-04-05:
Fixed in #83

closed 2023-04-05 by kasperkarlsson