CaringCaribou/caringcaribou #90: 报错

changcaixuezhang, 2023-06-26:
Traceback (most recent call last):
  File "/usr/local/bin/cc.py", line 33, in <module>
    sys.exit(load_entry_point('caringcaribou==0.4', 'console_scripts', 'cc.py')())
  File "/usr/local/lib/python3.10/dist-packages/caringcaribou-0.4-py3.10.egg/caringcaribou/caringcaribou.py", line 116, in main
    cc_mod.module_main(args.module_args)
  File "/usr/local/lib/python3.10/dist-packages/caringcaribou-0.4-py3.10.egg/caringcaribou/modules/uds.py", line 959, in module_main
    args.func(args)
  File "/usr/local/lib/python3.10/dist-packages/caringcaribou-0.4-py3.10.egg/caringcaribou/modules/uds.py", line 292, in __uds_discovery_wrapper
    arb_id_pairs = uds_discovery(min_id, max_id, blacklist,
  File "/usr/local/lib/python3.10/dist-packages/caringcaribou-0.4-py3.10.egg/caringcaribou/modules/uds.py", line 172, in uds_discovery
    with IsoTp(None, None) as tp:
  File "/usr/local/lib/python3.10/dist-packages/caringcaribou-0.4-py3.10.egg/caringcaribou/utils/iso15765_2.py", line 41, in __init__
    self.bus = can.Bus(DEFAULT_INTERFACE)
  File "/usr/local/lib/python3.10/dist-packages/python_can-4.2.2-py3.10.egg/can/util.py", line 369, in wrapper
    return f(*args, **kwargs)
  File "/usr/local/lib/python3.10/dist-packages/python_can-4.2.2-py3.10.egg/can/interface.py", line 124, in __new__
    kwargs = util.load_config(config=kwargs, context=config_context)
  File "/usr/local/lib/python3.10/dist-packages/python_can-4.2.2-py3.10.egg/can/util.py", line 207, in load_config
    bus_config = _create_bus_config(config)
  File "/usr/local/lib/python3.10/dist-packages/python_can-4.2.2-py3.10.egg/can/util.py", line 226, in _create_bus_config
    raise CanInterfaceNotImplementedError(
can.exceptions.CanInterfaceNotImplementedError: Unknown interface type "None"

kasperkarlsson, 2023-06-27:
Hello.

We will need some more context than just a stack trace. For starters:

- Have you installed Caring Caribou according to the instructions?
- Is the device/interface specified through the `-i <name>` argument available?
- What were you trying to do when this happened? Which command was run?

The error you get is documented in https://python-can.readthedocs.io/en/stable/errors.html#can.exceptions.CanInitializationError

changcaixuezhang, 2023-06-28:
> 你好。
> 
> 我们需要更多的上下文而不仅仅是堆栈跟踪。对于初学者：
> 
> * 您是否按照说明安装了 Caring Caribou？
> * 通过参数指定的设备/接口是否`-i <name>`可用？
> * 当这件事发生时你想做什么？运行了哪个命令？
> 
> 您收到的错误记录在https://python-can.readthedocs.io/en/stable/errors.html#can.exceptions.CanInitializationError

Ubuntu版本22.04ARM64   python3.10   安装步骤按照所给步骤进行操作，连上pcan可以接收到ecu的报文，但是在输入下面命令时出现以上错误:cc.py -i can0 uds discovery

kasperkarlsson, 2023-07-01:
I suspect the interface/channel configuration file for Python-can is missing. An easy fix for this can be found in https://github.com/CaringCaribou/caringcaribou/blob/master/documentation/troubleshooting.md#missing-default-configuration-file-for-python-can

This is also described in the installation instructions for Caring Caribou: https://github.com/CaringCaribou/caringcaribou/blob/master/documentation/howtoinstall.md#configure

closed 2023-07-08 by kasperkarlsson