CaringCaribou/caringcaribou #104: expose CAN FD support

droogie, 2024-02-02:
Consider exposing CAN FD support as an option, I quickly added it locally by changing the following:

```diff
diff --git a/caringcaribou/utils/can_actions.py b/caringcaribou/utils/can_actions.py
index 970fc1f..6ce3a39 100644
--- a/caringcaribou/utils/can_actions.py
+++ b/caringcaribou/utils/can_actions.py
@@ -110,8 +110,10 @@ class CanActions:
         self.add_listener(listener)
 
     def send(self, data, arb_id=None, is_extended=None, is_error=False, is_remote=False):
+        is_fd = False
         if len(data) > 8:
-            raise IndexError("Invalid CAN message length: {0}".format(len(data)))
+            is_fd = True
+            #raise IndexError("Invalid CAN message length: {0}".format(len(data)))
         # Fallback to default arbitration ID (self.arb_id) if no other ID is specified
         if arb_id is None:
             if self.arb_id is None:
@@ -124,7 +126,8 @@ class CanActions:
                           data=data,
                           is_extended_id=is_extended,
                           is_error_frame=is_error,
-                          is_remote_frame=is_remote)
+                          is_remote_frame=is_remote,
+                          is_fd=is_fd)
         self.bus.send(msg)
 
     def bruteforce_arbitration_id(self, data, callback, min_id, max_id,
```

and then adding this to my .canrc
```
fd = True
```

kasperkarlsson, 2024-02-05:
Hi there @droogie and thanks for posting this issue!

Your code looks like a really simple way of adding a nice feature 👍 If you want to get listed as a project contributor here on GitHub, feel free to send a PR with the above change. Otherwise I can of course implement your change on my own 🙂 The choice is yours!

kasperkarlsson, 2024-02-12:
Implemented in #105

closed 2024-02-12 by kasperkarlsson