CaringCaribou/caringcaribou #106: Caringcaribou not working with CAN-FD, Extended CAN protocol

Manishpentester219, 2024-02-15:
Hi caringcaribou team 
Need a new update for caringcaribou where CAN-FD and Extended CAN protocol support.

kasperkarlsson, 2024-02-16:
Hello,

You can enable FD support in `python-can`  by adding `fd = True` to your `.canrc` file, or by setting it in an environment variable. The `python-can` documentation can be found at <https://python-can.readthedocs.io/en/stable/configuration.html> and another user mentions their setup in #104 - this suggestion was recently implemented, so basic support for CAN FD messages is now exposed by the send function.

As for implementing FD modules, any contributions are of course welcome.

Manishpentester219, 2024-02-16:
CAN FD message is being sent, actually CAN FD has Extended Arbitration ID
due to it  XCP Module, UDS Module etc are not working. If you have any
solution regarding it please explain me in brief!

Thanks& Regards
Manish Chaudhary
Automotive security consultant

kasperkarlsson, 2024-02-19:
As mentioned in the previous comment, the current state of CAN FD support in Caring Caribou is that basic support for CAN FD messages was recently implemented in the send function.

This is not yet used by the built-in modules, but should give you a starting point when implementing modules suiting your needs. Any contributions adding such capability to the tool (or technical discussions on how to implement this in specific modules) are of course welcome.

Manishpentester219, 2024-02-21:
Sure ,
First add below things in all module
1.support extended arbitration I'd (0x12345678)

2.DLC (8,12,16,24)

3.Support Message length is upto 64 bit

After that send me updated tools , I'll check and update you  , your tool
is working or not .

Thanks & Regards
Manish Chaudhary
Automotive security consultant

kasperkarlsson, 2024-02-21:
Again - you are free to fork the project and implement the features that you need, and you are most welcome to send a pull request if you want to share your enhancements afterwards.

Simply placing orders will unfortunately not get them implemented.

closed 2024-02-21 by kasperkarlsson