CaringCaribou/caringcaribou #110: No Response In Doip Module

mwmarry, 2024-07-03:
I tried to test the ecu using the doip module, but all the submodules said "Power cycle your ECU and wait for a few seconds for the broadcast to be received"

I tried to restart the device, but nothing worked. I tried to capture packets using wireshark and could see that some identifucation packets existed.
[image]

So in what scenarios do we need to use the doip module? Thanks!

kasperkarlsson, 2024-07-08:
Perhaps @Cr0wTom can shed some light on this? 🙂

Cr0wTom, 2024-07-09:
Indeed from your screenshot I can see that there is a DoIP vehicle announcement message. 

The module is developed with the DoIP client library(https://pypi.org/project/doipclient/), and basically expects the Vehicle Identification Announcement Broadcast message to discover the ECU on the network. Commonly this is happening when the ECU boots. The problem is that this is automotive, and not everything is as streamlined as expected. That's also the reason on why this tool (CC in general) is not the perfect one button solution. Manufacturers don't strickly follow the standards (which standards have their own issues), and this results in issues like this. 

It needs further investigation and understanding of your setup to get to the end of it.

Also, something that has to be mentioned, is that from the development of the module till now, I had several DoIP targets in my hands, which prooved to me the limitations of the library. I had severe issues with communcation over IPv6, and as I said if the target is not following the standard 100%, the library falls short really quickly. 

**TLDR**; The module is just a wrapper of the doip client, with ultimate goal to perform UDS testing over DoIP, but the limitations of doip client library, with the fact that OEM/supplier didn't follow the standard 100% can lead to the module not working properly.

Cr0wTom, 2024-07-09:
Just to wrap it up, @kasperkarlsson I don't believe that this is an "issue" of CC, as the module is based on doipclient library. This is either a setup issue from @mwmarry side, or a library issue as the functionality here is really simple: 

[image]

kasperkarlsson, 2024-07-09:
Roger that - thank you very much for this thorough explanation 💪 I will leave this open just a bit more to give @mwmarry a chance to supply additional feedback before closing the issue.

kasperkarlsson, 2024-07-14:
Seeing as we haven't received any further comments or questions, I consider this issue resolved 👍

closed 2024-07-14 by kasperkarlsson