CaringCaribou/caringcaribou #112: Failed to receive CANFD packets!

sut4nd0, 2024-07-05:
Failed to receive CANFD packets!

kasperkarlsson, 2024-07-08:
Without any further details, this will be hard to look into.

sut4nd0, 2024-07-09:
By reading the code, I found the problem. When connecting CANFD, I need to specify like: `bus = can.interface.Bus(channel='can0', interface='socketcan', fd=True, bitrate=500000, dbitrate=2000000)`.
I think the most important of these is `fd=True, bitrate=500000, dbitrate=2000000`.

sut4nd0, 2024-07-09:
> Without any further details, this will be hard to look into.

By reading the code, I found the problem. When connecting CANFD, I need to specify like: `bus = can.interface.Bus(channel='can0', interface='socketcan', fd=True, bitrate=500000, dbitrate=2000000)`.
I think the most important of these is `fd=True, bitrate=500000, dbitrate=2000000`.

kasperkarlsson, 2024-07-09:
I am happy to hear that you managed to resolve your issue. Thanks for letting me know!

closed 2024-07-09 by kasperkarlsson