CaringCaribou/caringcaribou #115: Add decoded NRC text to negative response messages

alexdetrano, 2024-07-15:
See https://github.com/CaringCaribou/caringcaribou/pull/114/commits/8cf8f6528cd60262db5707927cd17809044d0604

The gist of it is that anytime the ECU responds with a negative response code (NRC), we should inform the user what that code is and what the human readable version of that code is. For example an NRC of 0x33 means "Security Access Denied". Currently there is some code only for the ReadMemoryByAddress function, but I'm filing this issue as an enhancement so that all UDS functions will respond with similar messages.

the following changes should be made:

* Create a function like nrc_to_text(nrc) that returns the text version of an NRC
* create a function like print_nrc_error_msg(nrc) something like
```
def print_nrc_msg(nrc):
    nrc_desc = nrc_to_text(nrc)
    return (f"Negative Response Code (NRC): {status:02x} - {nrc_descr}")
````
* then anywhere in the code where we inform the user of an NRC, we call the `print_nrc_msg` function

kasperkarlsson, 2024-07-16:
Hi there - yesterday I missed that you created a separate issue for this enhancement suggestion (great that you did this by the way - my bad that I didn't notice until now), so I responded in https://github.com/CaringCaribou/caringcaribou/pull/114#issuecomment-2229738119 🙂

There is already a lookup dict, which can be used as `NRC_NAMES.get(nrc)` (corresponding to your suggested `nrc_to_text(nrc)`).  Implementing an additional function which also prints a message containing the NRC name sounds like a good idea, I'll try it out and reach back!

kasperkarlsson, 2024-07-16:
I implemented a wrapper function `get_negative_response_code_name(nrc)` as you suggested, which returns the NRC name for a given NRC value. Your other suggestion regarding the print function was implemented as `print_negative_response_code(nrc)`. I also made sure these are used consistently throughout the module, rather than NRC lookup being spread out across several similar implementations and having to specify default values explicitly everywhere 😅

Feel free to have a look at the branch https://github.com/CaringCaribou/caringcaribou/tree/uds-nrc-print-function (diff here https://github.com/CaringCaribou/caringcaribou/compare/master...uds-nrc-print-function ) and let me know what you think! Once we are satisfied I'll make a PR and merge it into master.

kasperkarlsson, 2024-08-05:
Implemented in #119 👍

closed 2024-08-05 by kasperkarlsson