CaringCaribou/caringcaribou #123: uds_fuzz not working

Rivlen, 2024-08-22:
When attempting to use the uds_fuzz I get this error message:
```
asw@WS-5CD1166ST7:~/projects/caringcaribou$ caringcaribou uds_fuzz -h

-------------------
CARING CARIBOU v0.7 - python 3.10.12 (main, Jul 29 2024, 16:56:48) [GCC 11.4.0]
-------------------

Loading module 'uds_fuzz'

Traceback (most recent call last):
  File "/usr/local/bin/caringcaribou", line 33, in <module>
    sys.exit(load_entry_point('caringcaribou==0.7', 'console_scripts', 'caringcaribou')())
  File "/usr/local/lib/python3.10/dist-packages/caringcaribou-0.7-py3.10.egg/caringcaribou/caringcaribou.py", line 134, in main
    cc_mod = load_module(args.module)
  File "/usr/local/lib/python3.10/dist-packages/caringcaribou-0.7-py3.10.egg/caringcaribou/caringcaribou.py", line 116, in load_module
    cc_mod = importlib.import_module(full_module_name)
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/usr/local/lib/python3.10/dist-packages/caringcaribou-0.7-py3.10.egg/caringcaribou/modules/uds_fuzz.py", line 3, in <module>
    from caringcaribou.modules.uds import ecu_reset, print_negative_response, request_seed, extended_session
ImportError: cannot import name 'print_negative_response' from 'caringcaribou.modules.uds' (/usr/local/lib/python3.10/dist-packages/caringcaribou-0.7-py3.10.egg/caringcaribou/modules/uds.py)
```

kasperkarlsson, 2024-08-22:
Looks like this was missed in https://github.com/CaringCaribou/caringcaribou/pull/119 - thanks for reporting!

kasperkarlsson, 2024-08-25:
Hi @Rivlen,

The issue has been resolved in #125. Can you please check whether this works for you?

Rivlen, 2024-08-26:
It now works, thanks!

kasperkarlsson, 2024-08-26:
Great - and thank you for both reporting the issue and confirming the patch 👍

closed 2024-08-26 by kasperkarlsson