CaringCaribou/caringcaribou #124: cc.py or caringcaribou?

UnaDirezione, 2024-08-22:
Could it be due to the version update? Previously, I could use `cc.py` to execute, but now it seems to be unsupported.

UnaDirezione, 2024-08-22:
i find it! its in tags xxx sy i download the master

kasperkarlsson, 2024-08-22:
Once Caring Caribou is installed (as described in https://github.com/CaringCaribou/caringcaribou?tab=readme-ov-file#get-started ), the tool can be run using either `caringcaribou` (preferred) or `cc.py` (kept for backwards compatibility). The implementation of this can be seen in https://github.com/CaringCaribou/caringcaribou/blob/9c8e2d7ddfdf3ec79f197e30c5f30b49e13fdd40/setup.py#L39

Since the installation procedure was implemented, you can run it from any path instead of explicitly executing the script file.

Do you consider your issue resolved?

UnaDirezione, 2024-08-22:
> Once Caring Caribou is installed (as described in https://github.com/CaringCaribou/caringcaribou?tab=readme-ov-file#get-started ), the tool can be run using either `caringcaribou` (preferred) or [...]

thx

closed 2024-08-22 by kasperkarlsson