CaringCaribou/caringcaribou #128: When the parameters for doip diagnosis are not extended, the maximum logical address of the client can only be up to 0x7ff. Should it be 0x0exx?

xz0x, 2024-09-03:


kasperkarlsson, 2024-09-03:
Hello,

The CAN base frame format includes an 11 bit arbitration ID value, whose possible range is hence 0x0-0x7FF.

I am not quite sure how to parse `0x0exx` (as `x` is not a valid hexadecimal digit) and why this should be the maximum value. Can you please guide me through your reasoning here? 🙂

xz0x, 2024-09-03:
Hello, you're right. What I mean is diagnosis based on DOIP, not on CANbus, such as in Wireshark,
[image]
This is a session control for a 0x10 service, which requires route activation before proceeding。 In doip.py, the address of the external testing instrument is determined by two bytes, using a variable called send_arb_id. However, this is only up to 0x7ff, so that's what I meant
[image]

xz0x, 2024-09-03:
xx represents any data

n135c10r, 2025-06-12:
@kasperkarlsson It seems that xz0x is correct. According to the documentation address range 0x0E00–0x0FFF is allocated for clients / diagnostic equipment.

[image]

I wonder how strictly OEMs enforce this addressing in real-world DoIP implementations.
I need to look for an DoIP interface and some ECU on which I can do tests / research to familiarize more with this protocol.