CaringCaribou/caringcaribou #129: CANFD support

UnaDirezione, 2024-09-25:
How can I configure the code so that I can use cc. py on a bus that only supports CANFD/CANFD acceleration

kasperkarlsson, 2024-09-25:
Please refer to the configuration instructions for python-can:
<https://python-can.readthedocs.io/en/stable/configuration.html>

I suspect it might be as easy as adding an `fd = true` entry to your `~/.canrc` file. If this is not enough you might want to look into setting the `is_fd` parameter to `true` in calls to underlying python-can functions (such as the wrapper in CanActions.send, see <https://github.com/CaringCaribou/caringcaribou/blob/master/caringcaribou/utils/can_actions.py#L107>)

If you get this working, please share your method here to help others who are in a similar situation 🙂

sut4nd0, 2024-10-21:
when configuring "can.Bus()", add "fd=True". And, When constructing "can.Message", set "is_fd" to "True".