CaringCaribou/caringcaribou #135: How to configure a PEAK PCAN PCI CAN interface with pcanbasic driver

Atm-Acn, 2025-06-18:
My test harness assembles a PEAK PCAN PCI interface instead of a USB interface. Also, for legacy reasons, we use the pcanbasic driver instead of SocketCAN.
OS: Ubuntu 18.04.6 LTS (GNU/Linux 4.15.0-176-generic x86_64)
Driver version: Release_20220124_n
It is still possible to setup the CAN interface? Which modifications are needed?

n135c10r, 2025-06-18:
CaringCaribou relies on the python‑can library for CAN interface support. You should follow the instructions for correctly configuring your interface here: https://python-can.readthedocs.io/en/stable/interfaces/pcan.html

So, if your driver is installed correctly, you should just modify your .canrc file with these settings:

[default]
interface = pcan
channel   = PCAN_PCIBUS1
bitrate   = 500000

And the CaringCaribou should work normal.

kasperkarlsson, 2025-06-18:
Agreed - what @n135c10r suggests above should resolve your issues.

Atm-Acn, 2025-06-18:
It works, thanks @n135c10r

closed 2025-06-18 by Atm-Acn