CaringCaribou/caringcaribou #136: UDS Discovery Issue

safwannasser, 2025-08-15:
Hi,

If there are any periodic CAN messages sending 0x50 in its 1st byte, the UDS discovery functoin catches all those as a valid server ID, so I'm getting the server ID for all client requests. Whats a workaround for this?
Thanks in advance

kasperkarlsson, 2025-08-21:
Hello @safwannasser and thank you for reaching out!

A verification step was previously added to prevent accidental matches - after all, the [`is_valid_response(message)`](https://github.com/CaringCaribou/caringcaribou/blob/master/caringcaribou/modules/uds.py#L212) function is quite likely to give false positives on CAN buses with high traffic.

Just as you have already figured out, the `--blacklist B [B ...]`/`-b B [B ...]` option can be used to remove any known IDs. There is also an option `--autoblacklist N`/`-ab N` which can be used to listen for false positives to automatically blacklist before running the discovery. For example, `caringcaribou uds discovery -autoblacklist 10` will listen for 10 seconds and blacklist any matches during this period, before initiating the discovery.

As for the 11 vs 29 bits mismatch, I do believe that this might warrant a fix. This could be implemented via an optional argument to apply stricter filtering based on arbitration ID length. Can you please supply an example of the command you currently run and a few CAN messages resulting in a false positives? Just to give me an example to test for during development 🙂

kasperkarlsson, 2025-08-26:
Thank you very much for these clarifications, and great job finding both of these bugs! 

You are of course welcome to send a PR with these fixes! Otherwise I can of course implement them on my own, but I would rather let you get the credit 🙂